velvet-os/imagebuilder #307: kukui 230917-01 on debian 12: tpm0 missing/gone

kxtzownsu, 2025-02-28:
On a kukui debian 12 230917-01 image, I'm unable to access /dev/tpm0 (not found / missing) and can't do TPM-related stuff that I need to do. 

Side note: the TPM works just fine within CrOS so its not chip failure. 

My idea to the TPM not working is the bootloader but I'm unable to prove that at this time.

thenameisluk, 2025-02-28:
@kxtzownsu 
my guess would be it's not in mainline kernel or dts, basically a chrome os kernel exclusive for now

what exactly is ur tpm related stuff?
you can still encrypt your partitions (even root one) with password without tpm if that's what you need

thenameisluk, 2025-02-28:
@kxtzownsu could you check if it's available on the new testing images?
https://github.com/hexdump0815/imagebuilder/discussions/302

kxtzownsu, 2025-03-01:
@thenameisluk was able to test a few minutes and and seems to be working.

after more investigation the reason it was missing is due to missing kernel modules & firmware. for anyone on these devices looking at this issue, compiling the modules for `6.1.51-stb-mt8+` should allow you to use the TPM

kxtzownsu, 2025-03-01:
tpm stuff works on `6.1.51-stb-mt8+` with the kernel drivers

[image]