velvet-os/imagebuilder #311: chromebook_homestar: audio dosent work, neither lid sleep

votisek, 2025-03-13:
I have used my Duet 5 for a month or so, withou a problem, except speakers and a mic, but that was okay. I wa still tolerating it, but about a week ago my close-the-lid-to-sleep broke, i dont know why. Can you please help me? 
`uname`: `Linux developer 6.12.0-stb-cbq+ #15 SMP PREEMPT Mon Nov 18 08:55:27 CET 2024 aarch64 GNU/Linux`

thenameisluk, 2025-03-13:
@votisek for the audio just install the patch https://github.com/hexdump0815/imagebuilder/discussions/240#discussioncomment-10732006
and everything will work

lid sleep works for me on gnome and i daily drive this one so would likely notice
```
Linux chluk 6.12.0-stb-cbq+ #15 SMP PREEMPT Mon Nov 18 08:55:27 CET 2024 aarch64 GNU/Linux
```
so it might be a Desktop environment?

votisek, 2025-03-13:
I use gnome w/ caffeine and dash to dock, but i can try xfce

thenameisluk, 2025-03-13:
@votisek oh yeah trixie
gimme a moment

votisek, 2025-03-13:
alr, tried gnome classic and gnome, both even on xorg (dont know what that is) and tried the default desktop manager and xfce, will try installing sound patch

thenameisluk, 2025-03-13:
so yeah mutter on trixie has a bug where it kinda freezes for no reason
you might be experiencing that
and even (only on some versions) crashes when using osk

i would recommend downgrading and waiting for full gnome 48 release

votisek, 2025-03-13:
i have homestar and tried using xfce, a desktop manager. tried installing the path, but didnt help

thenameisluk, 2025-03-13:
if you are talking about audio patch
try running after installation
```
systemctl --user restart pipewire
```
and seeing if it works

votisek, 2025-03-13:
it unfortunately dosent work, in sound settings it still shows just dummy output, which cannot make a sound

thenameisluk, 2025-03-13:
and you have installed the patch from velvet repo?

votisek, 2025-03-13:
yeah, i used `sudo apt install homestar-audio-patch`, and it installed, so i hope i did...

thenameisluk, 2025-03-13:
that's weird is should have worked

thenameisluk, 2025-03-13:
could you try putting
```
# Use case configuration for ALC5682+MAX98357A on SC7180

SectionVerb {
	Value {
		TQ "HiFi"
	}
}

SectionDevice."Speaker" {
	Comment "Speaker"

	Value {
		PlaybackChannels 4
		PlaybackPCM "hw:${CardId},1"
		PlaybackPriority 100
		PlaybackMixerElem "Speaker"
		PlaybackVolume "Speaker Playback Volume"
	}
}

SectionDevice."Mic" {
	Comment "Internal Microphone"

	EnableSequence [
		cset "name='IF1 01 ADC Swap Mux' 1"
		cset "name='Stereo1 ADC L2 Mux' 1"
		cset "name='Stereo1 ADC R2 Mux' 1"
		cset "name='Stereo1 ADC MIXL ADC2 Switch' 1"
		cset "name='Stereo1 ADC MIXR ADC2 Switch' 1"
	]

	DisableSequence [
		cset "name='Stereo1 ADC L2 Mux' 0"
		cset "name='Stereo1 ADC R2 Mux' 0"
		cset "name='Stereo1 ADC MIXL ADC2 Switch' 0"
		cset "name='Stereo1 ADC MIXR ADC2 Switch' 0"
	]

	Value {
		CapturePCM "hw:${CardId},0"
		CapturePriority 100
		CaptureMixerElem "STO1 ADC"
		CaptureVolume "STO1 ADC Capture Volume"
	}
}

SectionDevice."HDMI" {
	Comment "HDMI Jack"
	
	ConflictingDevice [
		"Speaker"
	]

	EnableSequence [
	]

	DisableSequence [
	]

	Value {
		PlaybackPriority 200
		PlaybackPCM "hw:${CardId},2"
		JackControl "HDMI Jack"
	}
}
```
in ```/usr/share/alsa/ucm2/Qualcomm/sc7180/rt5682-max98357a/HiFi.conf```

and running ```systemctl --user restart pipewire```

thenameisluk, 2025-03-13:
what's your homestar rev?
```
cat /proc/device-tree/model
```

votisek, 2025-03-13:
rev4+

thenameisluk, 2025-03-13:
hmm 'm currently on 3
i don't know if my other homestar is 4+

but i don't think it would matter
gimme a moment

thenameisluk, 2025-03-13:
well i do have here fully working (audio and lid suspense) duet 5 rev4+
on gnome 47 (it's a bit dated since i don't use this device rn)
so the good news is this isn't a hardware issue

[image]

thenameisluk, 2025-03-13:
@votisek you can check if it works better on testing images : https://github.com/hexdump0815/imagebuilder/discussions/302
and if audio patch fixes stuff there

imma head to sleep, if it doesn't work i will try to help you when i wake up

votisek, 2025-03-13:
couldnt there be a problem in you using sid meanwhile me using bookworm?

[image]

thenameisluk, 2025-03-13:
> couldnt there be a problem in you using sid meanwhile me using bookworm?
might be, check out the testing images ^ these are close my system
i really need sleep thoe

votisek, 2025-03-13:
For sure, i need to sleep too. Good night

votisek, 2025-04-15:
Hey, wanted to come back to this issue after a long time. I moved to sway and Hyprland and that didnt solve the issue. Do you think that there is a chance reinstalling the system solves the issue? Or building it instead of downloading a prebuilt image? I am on spring break rn and hav alot of time. Thx alot

thenameisluk, 2025-04-16:
i'd probably try the testing image (without installing)
https://github.com/hexdump0815/imagebuilder/discussions/302#discussion-8004558
then apply the audio patch

if the audio works great, let me know and i'll try to guide you through the installation
since it requires modified version of procedure currently described in documentation

when it comes to suspense, it's iffy when booted from usb so i wouldn't worry about it when running before the installation

thenameisluk, 2025-04-18:
vtinstall is now in the repo

if you update velvettools to 0.7.6 you should have it

feedback would be appreciated

raremotas, 2025-07-02:
I have tried both the new and old build and I can't figure out audio.

It seems the image uses pulse audio by default? So I have installed pipewire pipewire-pulse.

I am able to get audio to come out of just the left speakers by changing the config to "pro audio" and selecting one of the options. It thinks its quad channel but it just comes out the left. I have verified the HiFi.conf is in the right spot too!

Any pointers can help especially I am confused about the pulse audio vs pipewire situation and if I am supposed to be intalling pipewire. Thanks so much for everything you've done for this device! :)

thenameisluk, 2025-07-02:
no problem 👍️

raremotas, 2025-07-02:
Looks like what was happening is I was only getting the Pro Audio configuration and not the Default (speakers, mic) one.

I now have installed to the internal storage using vtinstall! 

Unfortunately it looks like the audio wasn't fully fixed it is still only coming from two speakers it's just the top two now. When I swap to Pro Audio and select option two it comes out of the left two. I have verified that the HiFi.conf looks right.

Could installing gnome help somehow? Maybe something with my pipewire?

thenameisluk, 2025-07-02:
if the gnome works correctly
it might be pipewire issue

my guess
it isn't really designed to handle more than left and right speaker

i would try opening a bug report there or looking for similar issue with system that has 4 distinct speakers

thenameisluk, 2025-07-02:
@raremotas you probably should install ```gnome-core``` instead of just ```gnome``` you might be missing some deps

thenameisluk, 2025-07-04:
isn't pipewire meant for routing audio?

you should be able to make virtual device and route it the way you want

raremotas, 2025-07-07:
Yes that's right I found out there's a cute little audio patching program called Helvum which lets you route the speakers however you'd like on a per app basis. I swear they sound so much better than on ChromeOS.

I am attaching an image of the patching in Helvum you just route FR and FL from your program into RL and RR speakers. Libre Wolf is the app in this example and it routes to Built-In Audio Speaker.

[image]

Do you have any idea of how the touchpad could maybe gain natural scrolling and gestures under gnome? I could reverse the scroll direction in xfce but it didn't transfer over to gnome. Also have you been using gnome under Wayland or Xorg?

thenameisluk, 2025-07-10:
i think that's bug in debian trixie

on sid there is no such issue

raremotas, 2025-07-12:
I tried sid but sadly that didn't fix for me ;~;

 Imma try and full fresh reinstall and then just swap to sid sources then upgrade and install gnome

raremotas, 2026-03-22:
Looks like that gnome issue ironed itself out over time like u mentioned.

ANYWAYS GUESS WHAT! Idk why this took me so long tbh. I was able to finally cleanly replicate the speaker config from ChromeOS.

make this file and edit it:
`nano ~/.config/pipewire/pipewire.conf.d/quad-upmix.conf`

These are the contents to put in the file. This creates an audio sink which upmixes the FR, RR, FL, RL into stereo on the quad speakers. Conceivably this could be further improved through a script which swaps sinks when the screen is rotated. This is for landscape.

```
context.modules = [
  { name = libpipewire-module-loopback
    args = {
      node.description = "Quad Upmix"

      audio.position = [ FL FR ]

      capture.props = {
        node.name = "quad_upmix_in"
        media.class = "Audio/Sink"
      }

      playback.props = {
        node.name = "quad_upmix_out"
        target.object = "alsa_output.platform-sound.HiFi__Speaker__sink"
        audio.position = [ FL FR RL RR ]
        node.passive = true
        stream.dont-remix = true
        channelmix.upmix = true
        channelmix.upmix-method = simple
      }
    }
  }
]

```

restart audio:
`systemctl --user restart pipewire pipewire-pulse wireplumber`

You gotta hop in the gnome audio settings and swap the output device to "Quad Upmix" but you can call it something different too. Let me know if this works for you! Thanks.