velvet-os/imagebuilder #312: chromebook_corsola binfmt dosent work?

LukeDevsE, 2025-03-15:
on startup, it gives an error saying around
failed to mount binfmt-support
and a bunch of other binfmt errors.
when using binfmt in the terminal when it boots up, doesn't work either, saying it may have failed mounting binfmt when starting up.

thenameisluk, 2025-03-15:
could you check if issue is till present on new testing images?
https://github.com/hexdump0815/imagebuilder/discussions/302#discussion-8004558

LukeDevsE, 2025-03-15:
so i installed the new testing image, and i tried pi apps to install steam (which requires binfmt)
but pi apps only works on bookworm, anyway i can use bookworm apts on trixie?

thenameisluk, 2025-03-15:
well in theory you can
but there is a chance for dependency issue

LukeDevsE, 2025-03-15:
is there any way i could downgrade the test image to bookworm?

thenameisluk, 2025-03-15:
without breaking ur system no simple one

if all you want is steam, you can try [this guide](https://wiki.postmarketos.org/wiki/Steam_in_FEX)
just replace apk commands with apt ones

still unless you want to play old 2d games it won't perform well
you have to take into account that this device doesn't support vulkan
there is only opengl 3.1 or lower