velvet-os/imagebuilder #314: chromebook_corsola test image explodes if you update

LukeDevsE, 2025-03-18:
just clicked update my system in power off menu in gnome
[image]

LukeDevsE, 2025-03-18:
when logging in, it kinda looks like a bunch of glitched out squares of the screen

LukeDevsE, 2025-03-18:
terminal looks perfectly fine

LukeDevsE, 2025-03-18:
when i was updating, my screen went fully black and i pressed the power button to turn on (i didnt hold it though to force turn off)

LukeDevsE, 2025-03-18:
just in case anyone asks, its not a gpu issue chrome os displays perfectly fine

LukeDevsE, 2025-03-18:
anyways i guess im just gonna reinstall velvetos and just use kde this time

LukeDevsE, 2025-03-18:
ok i am very confused now, with the EXACT same sd card, im pretty sure i forced turned it off on accident during startup. and now its doing the same exact thing

LukeDevsE, 2025-03-18:
but the weird thing is, it also did a random shutdown like what it did earlier

LukeDevsE, 2025-03-18:
as soon as i pressed the turn off button for a split second it was all normal?

LukeDevsE, 2025-03-18:
so for some reason, in kde, when you press the power button it goes back to normal, if you type in your password, it goes back to normal its so confusing

LukeDevsE, 2025-03-18:
im going to use postmarketos again for now, until i can figure out how to stop this from happening

thenameisluk, 2025-03-18:
that's a lot of text

imma try the image on my corsola to figure out what's going on

@LukeDevsE  just to be sure we are talking about ideapad slim 3?

thenameisluk, 2025-03-18:
i can confirm the issue exists

will try to figure out what's happening

thenameisluk, 2025-03-18:
@hexdump0815 there appears to be a lot of
```
[  371.236839] mtk_iommu_isr: 497495 callbacks suppressed
[  371.236854] mtk-iommu 14016000.iommu: fault type=0x5 iova=0xfe96f000 pa=0x0 master=0x5000008(larb=0 port=2) layer=0 read
[  371.236878] mtk-iommu 14016000.iommu: fault type=0x5 iova=0xfe9c9000 pa=0x0 master=0x5000008(larb=0 port=2) layer=0 read
[  371.236893] mtk-iommu 14016000.iommu: fault type=0x5 iova=0xfe9d8000 pa=0x0 master=0x5000008(larb=0 port=2) layer=0 read
[  371.236904] mtk-iommu 14016000.iommu: fault type=0x5 iova=0xfe9e7000 pa=0x0 master=0x5000008(larb=0 port=2) layer=0 read
[  371.236914] mtk-iommu 14016000.iommu: fault type=0x5 iova=0xfe9e8000 pa=0x0 master=0x5000008(larb=0 port=2) layer=0 read
[  371.236925] mtk-iommu 14016000.iommu: fault type=0x5 iova=0xfe9f7000 pa=0x0 master=0x5000008(larb=0 port=2) layer=0 read
[  371.236937] mtk-iommu 14016000.iommu: fault type=0x5 iova=0xfea05000 pa=0x0 master=0x5000008(larb=0 port=2) layer=0 read
[  371.236947] mtk-iommu 14016000.iommu: fault type=0x5 iova=0xfea14000 pa=0x0 master=0x5000008(larb=0 port=2) layer=0 read
[  371.236959] mtk-iommu 14016000.iommu: fault type=0x5 iova=0xfea15000 pa=0x0 master=0x5000008(larb=0 port=2) layer=0 read
[  371.236969] mtk-iommu 14016000.iommu: fault type=0x5 iova=0xfea24000 pa=0x0 master=0x5000008(larb=0 port=2) layer=0 read
[  371.798563] rfkill: input handler enabled
[  375.227627] usb 2-1.2: USB disconnect, device number 3
```
in dmesg for some reason and graphics gets distorted as shown above

full dump:
[dump.txt](https://github.com/user-attachments/files/19316958/dump.txt)

thenameisluk, 2025-03-18:
the issue is related to mutter both x11 and wayland
gdm3 and gnome

there is no problem with lightdm nor xfce

hard to tell what's causing the issue

but i will try to reproduce it on other mtk chromebooks later

thenameisluk, 2025-03-18:
the issue seams to be mutter specyfic

not present on kde

once i verify if it's related to mali gpu
or corsola specyfic
imma try openning an issue upstream

thenameisluk, 2025-03-18:
the issue is also present on kukui (krane)
and it seams to be related to gtk
this time it was observed on phoc ("gnome mobile greater")
https://gitlab.gnome.org/World/Phosh/phoc

thenameisluk, 2025-03-18:
also it's not kernel related
since the kukui is still running 6.11

thenameisluk, 2025-03-18:
> GDK_GL_DISABLE=base-instance

this one doesn't look remotely related but will give it a shot

> playing around with v6.14
also if it was kernel related there wouldn't be an issue on krane with 6.11.0-stb-mt8+ which noone had any issues with
including me

it had to be something introduced in mesa or gtk for that got into debian testing lately
and has to be related to mali driver since there is no issue on qcom chromebooks

thenameisluk, 2025-03-18:
> shall we maybe create a "well known problem" page next to the faq one where we collect such things?

hard to tell if it will be fixed soon or not
assuming it plagues all or most mali devices it should be quickly spotted and fixed
so imma be hopeful for now and try to gather some more information

LukeDevsE, 2025-03-18:
i have a cm14 chromebook to be clear

thenameisluk, 2025-03-18:
doesn't matter at this point it seams to affect ALL chromebooks with mali graphics card

debian testing migrated to mesa 25 about 5 days ago
this might be the root cause or at least part of it

LukeDevsE, 2025-03-18:
does using lightdm on kde fix this issue?

thenameisluk, 2025-03-18:
on X11 yes

wayland i'm not sure

LukeDevsE, 2025-03-18:
i did use gdm (not lightdm) on both kde, and gnome, and they both had the same glitched out screen after a little while

thenameisluk, 2025-03-18:
@LukeDevsE even on x11?

LukeDevsE, 2025-03-18:
> [@LukeDevsE](https://github.com/LukeDevsE) even on x11?

i think i used wayland for both, i didnt really change it

thenameisluk, 2025-03-18:
x11 seams to work fine

LukeDevsE, 2025-03-18:
is there anyway where i can force the glitches to happen?

thenameisluk, 2025-03-18:
@hexdump0815 this is 100% mesa 25 and unrelated to gnome

LukeDevsE reported it broken on kde and gnome wayland
i got it on gnome, phog/phrog and hyperland wayland

x11 seams to be fine for now

hexdump0815, 2025-03-18:
@thenameisluk - this smells a bit like what is mentioned here as well: https://oftc.irclog.whitequark.org/linux-cros-arm/2025-03-18#34104406 ... so maybe lets wait for a few mesa updates to hopefully fix this ...

thenameisluk, 2025-03-18:
let's leave it to them then

thenameisluk, 2025-03-20:
i can confirm the issue is known upsteam https://gitlab.freedesktop.org/mesa/mesa/-/issues/12718
and has been discussed https://oftc.irclog.whitequark.org/panfrost/2025-03-05

there is a possible temporary work around
by downgrading mesa to bookworn backport version
https://packages.debian.org/pl/source/bookworm-backports/mesa
the issue should disappear
when the issue is addressed ppl should be save to upgrade again

tobhe, 2025-03-20:
I manually installed [libEGL-mesa from the staging/25.0 branch](https://gitlab.freedesktop.org/mesa/mesa/-/tree/staging/25.0?ref_type=heads) and that seems to fix it too. As far as I understand there is a new 25.0.2 point release due today so this will hopefully be fixed in Debian/Ubuntu soon.

thenameisluk, 2025-03-20:
not sure when it's going to be out
just in case there is a temporary fix
```bash
#get older mesa
apt build-dep mesa
git clone https://salsa.debian.org/xorg-team/lib/mesa -b mesa-24.3.4-3 --depth=1
cd mesa

#build it (this will take a while)
dpkg-buildpackage --build=any,all
cd ..
rm *dev*.deb
rm *dbgsym*.deb

#install it
sudo apt install ./*.deb
```

tobhe, 2025-05-05:
So it turns out the update in Debian did not in fact fix the issue. It seems like postmarket os is hitting it too https://gitlab.postmarketos.org/postmarketOS/pmaports/-/merge_requests/6470

closed 2025-05-21 by hexdump0815