velvet-os/imagebuilder #326: idea: maybe use systemd service to take care of display and touchscreen rotation

hexdump0815, 2025-05-06:
currently for some systems with portrait mode display panels display rotation is enabled via some rc.local hack on first boot for instance on krane. it might be a cleaner approach to do this via some systemd service as suggested here: https://github.com/hexdump0815/imagebuilder/discussions/284#discussioncomment-13003738

this github issue here is to potentially discuss this topic further and to take it out of that monster discussion there

thenameisluk, 2025-05-06:
that sounds like a great idea!!!

we might even make is general purpose like ```first-boot-setup.service```
which would be shared across system and would run ```/scripts/first-boot-setup.sh```

that could apply device specyfic fixes like
- above mentioned display/touchscreen rotation
- or [homestar audio patch](https://github.com/hexdump0815/imagebuilder/discussions/240#discussioncomment-10690792) (which conflicts with lazor so it can't be applied to all images)

archisman-panigrahi, 2025-05-10:
I wrote a GUI app for manually rotating the screen on X11 based setups. Works in my Lenovo 300e (hana/oak) chromebook.

Using the display settings only rotated the display, but not the touch input. This app also rotates the touch input.

https://github.com/archisman-panigrahi/surface-RT-screen-rotator

Not all chromebooks are powerful enough to run wayland based DEs (especially because of lack of graphics drivers). In such cases, this app helps a lot to rotate screen in X11 based DEs (e.g. XFCE/MATE). 

It maybe beneficial for the users if this is added to the readme and/or include this app in the default installation.

thenameisluk, 2025-05-10:
while the app looks nice

the "correct" rotation should be automatic

while still letting ppl adjust it if needed (with touch rotation as well) sounds like a good idea
it seams a bit outside of the scope of **this** issue

i think this might have its place in https://github.com/velvet-os/velvet-os.github.io/blob/main/postinst/readme.md
so if you want feel free to open a pr

also if you want we can add this as a package to [velvet repo](https://gitlab.com/velvet-os/velvet-repo) which is include in images by default since testing images

archisman-panigrahi, 2025-05-10:
> but the \"correct\" rotation should be automatic

In regular amd64 tablets, that is a wayland-only feature (requires iio-sensors-proxy), but it doesn't work with x11.

However, in the low end chromebooks, the user can only run xfce/mate, which are x11-only.

I will turn my repo into a debian package and send a PR.
Thank you