velvet-os/imagebuilder #329: homestar: Screen Rotation stopped working

jonascant, 2025-05-08:
Running Debian Trixie with Kernel 6.12 initially the Screen Rotation worked fine. I guess in the last few days an update broke automatic Screen Rotation.

Screen Rotation initially only worked with the keyboard detached. Maybe the tablet mode detection is not working anymore?

jonascant, 2025-05-08:
this seems to be a debian bug. Reinstalled once again, upgraded the kernel to 6.12 but stayed on bookworm. Rotation works.

thenameisluk, 2025-05-09:
when it comes to switching to tablet mode the cros_ec_button should be sending event to switch

[image]

if you have the issue with rotation being of 90 or 180

you might want to play around with matrix

https://github.com/hexdump0815/imagebuilder/issues/202#issuecomment-2002013535

jonascant, 2025-05-15:
I found the cause of my issue is the package iio-sensor-proxy. I downgraded it to 3.0-2 (Bookworm) and it works fine again.

hexdump0815, 2025-05-15:
@jonascant - it might be good to report that back to the package maintainer if possible

archisman-panigrahi, 2025-05-17:
@jonascant Could you add the touchscreen information of your Chromebook here - https://github.com/velvet-os/imagebuilder/issues/334 ?
We are trying to build an app for manual touch rotation for devices without a rotation center (that will also work for devices with a rotation sensor).

archisman-panigrahi, 2025-06-18:
The issue has been identified and patched upstream https://gitlab.freedesktop.org/hadess/iio-sensor-proxy/-/merge_requests/400#note_2965511. However, the patch may not appear in Debian Trixie as Trixie may be freezed before this new version is released.

I build the patched version in Launchpad. Please test and let me know if it works for you (for me, it worked in google-scarlet)

arm64: https://launchpad.net/~apandada1/+archive/ubuntu/iio-sensor-proxy-bufffix/+files/iio-sensor-proxy_3.7-6-buffix_arm64.deb
armhf: https://launchpad.net/~apandada1/+archive/ubuntu/iio-sensor-proxy-bufffix/+files/iio-sensor-proxy_3.7-6-buffix_armhf.deb

@thenameisluk Can we add this patched version to velvet os repo?

thenameisluk, 2025-06-19:
sorry but velvet repo is configured in such a way that it can't override debian packages
https://github.com/velvet-os/imagebuilder/blob/main/files/extra-files/etc/apt/preferences.d/velvet_repo.pref

archisman-panigrahi, 2025-06-19:
Can we override the pin priority of iio-sensors-proxy (and only that) with an update and ship iio-sensors-proxy in velvet-repo?

thenameisluk, 2025-06-19:
most we can do is ship separate package that patches the ```/usr/lib/udev/rules.d/80-iio-sensor-proxy.rules``` with post install script

thenameisluk, 2025-06-19:
this package is not part of velvet tools and it wouldn't be honest to ship it along with it

but like is they said that they are working on a fix

it should already be in trixie before we build an official image

and i'm like learning sewing so ... just take my [homestar audio fix](https://gitlab.com/velvet-os/velvet-repo/-/tree/master/source/packages/misc/trogdor/homestar-audio-patch-0.1.0_all?ref_type=heads) and modify it to patch the file it's not that hard

archisman-panigrahi, 2025-09-06:
This seems to have been fixed in Trixie https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1108066. Someone else please also check, and if it works, we can close the issue.