velvet-os/imagebuilder #330: What is the status of GPU support for mt8173?

archisman-panigrahi, 2025-05-11:
There are some relatively new commits in the kernel which are seemingly related to GPU support for mt8173 in Linux kernel (see commit message of https://github.com/torvalds/linux/commit/d7683c49638c83bcb6a817671bb348dc435da3b5).

I don't much about these drivers/kernel development to understand what it means.

Does this commit indicate that the GPU is supported in the upcoming Linux kernels, e.g. 6.15? Does it mean something else?

thenameisluk, 2025-05-11:
in theory it is already there https://gitlab.freedesktop.org/mesa/mesa/-/issues/12959#note_2894306

but noone here knows how to get it working

thenameisluk, 2025-05-11:
we already have discussion about that https://github.com/hexdump0815/imagebuilder/issues/270#issuecomment-2855243390

archisman-panigrahi, 2025-05-11:
Thank you for sharing

closed 2025-05-11 by archisman-panigrahi