velvet-os/imagebuilder #331: Homestar: How to compile Kernel

jonascant, 2025-05-15:
I couldn't find enough instructions for me to compile the kernel. I would like to add some modules (eg. for the apple magic trackpad). Is there a complete guide or script someone could point me to? I don't have experience in compiling my own kernel, so the more detail the better.

Thanks!

thenameisluk, 2025-05-15:
isn't https://github.com/hexdump0815/linux-mainline-and-mali-generic-stable-kernel/blob/master/readme.cbq
enough?

jonascant, 2025-05-15:
I found that before. Is this complete? Do I need to apply more patches for my specific device? (Duet 5)

thenameisluk, 2025-05-15:
why would you need more patches?
it already has all patches for ur system

jonascant, 2025-05-16:
I am afreight not. I really have not much experience. I pulled the mainline kernel from github and created the directories: ```# directories:
# - /compile/doc/stable - the files in this dir
# - /compile/source/linux-stable-cbq - the kernel sources checked out from gitrepo
# - /compile/result/stable - the resulting kernel, modules etc. tar.gz files
# - /compile/doc/kernel-config-options - https://github.com/hexdump0815/kernel-config-options
# - /compile/doc/kernel-extra-patches - https://github.com/hexdump0815/kernel-extra-patches
```
Now i don't know where to get the actual files i need and the patches