velvet-os/imagebuilder #334: the big chromebook touchscreen information collection

hexdump0815, 2025-05-16:
lets try to collect information about the touchscreens used in the various chromebooks as requested here: https://github.com/velvet-os/velvet-os.github.io/pull/9#issuecomment-2884237238 and here: https://github.com/velvet-os/imagebuilder/issues/182#issuecomment-2703928436 in this github issue ...

to me it looks like an "xinput list" output should be good for the first request to get the xinput ids and a script should be run as described here: https://github.com/linuxwacom/wacom-hid-descriptors/tree/master?tab=readme-ov-file#contributing for the second request - @archisman-panigrahi and @vulpes2 - please let me know any extra useful information about collecting this information which i should add to this github issue, so that people can easily collect the information and add it as comments to this github issue ... also if other touchscreen or maybe even touchpad related information should be collected, please let me know.

archisman-panigrahi, 2025-05-17:
Here is Chromebook `hana` (Lenovo 300e) -- oak family

```
$ xinput list
⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ hid-over-i2c 06CB:7817                  	id=8	[slave  pointer  (2)]
⎜   ↳ Elan Touchpad                           	id=9	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ gpio-keys                               	id=6	[slave  keyboard (3)]
    ↳ cros_ec                                 	id=7	[slave  keyboard (3)]
    ↳ EasyCamera: EasyCamera                  	id=10	[slave  keyboard (3)]
```

@hexdump0815 The command `xinput list` already collects the required touchpad related info (here it is `Elan Touchpad`)  for rotating the touchpad signal after rotating the screen. For my software (https://github.com/archisman-panigrahi/surface-RT-screen-rotator), this should enough. By the way, do you know how to know which Chromebook the user is using via some command? The command `sudo dmidecode -s system-product-name` does not produce any output.

archisman-panigrahi, 2025-05-17:
```
$ cat /proc/device-tree/model
Google Scarlet
```

```
xinput list
⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ hid-over-i2c 2D1F:0036 Stylus stylus    	id=6	[slave  pointer  (2)]
⎜   ↳ hid-over-i2c 2D1F:0036 stylus           	id=7	[slave  pointer  (2)]
⎜   ↳ Elan Touchscreen                        	id=8	[slave  pointer  (2)]
⎜   ↳ hid-over-i2c 2D1F:0036 Stylus eraser    	id=12	[slave  pointer  (2)]
⎜   ↳ hid-over-i2c 2D1F:0036 eraser           	id=13	[slave  pointer  (2)]
⎜   ↳ SEM USB Keyboard Consumer Control       	id=16	[slave  pointer  (2)]
⎜   ↳ Telink Wireless Receiver Consumer Control	id=18	[slave  pointer  (2)]
⎜   ↳ Telink Wireless Receiver Mouse          	id=20	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ cros_ec                                 	id=9	[slave  keyboard (3)]
    ↳ cros_ec_buttons                         	id=10	[slave  keyboard (3)]
    ↳ rk3399-gru-sound Headset Jack           	id=11	[slave  keyboard (3)]
    ↳ SEM USB Keyboard                        	id=14	[slave  keyboard (3)]
    ↳ SEM USB Keyboard System Control         	id=15	[slave  keyboard (3)]
    ↳ SEM USB Keyboard Consumer Control       	id=17	[slave  keyboard (3)]
    ↳ Telink Wireless Receiver Consumer Control	id=19	[slave  keyboard (3)]
    ↳ Telink Wireless Receiver                	id=21	[slave  keyboard (3)]
    ↳ Telink Wireless Receiver System Control 	id=22	[slave  keyboard (3)]
```

In this case, if I use `Elan Touchscreen` for the screen and `hid-over-i2c 2D1F:0036 Stylus stylus ` in the touchpad entry of Screen Orientation Manager, both the touch input and the stylus rotates perfectly.

WesleyVanNeck, 2025-05-18:
xinput list of the lenovo duat 3 (Wormdingler)

`cat /sys/firmware/devicetree/base/model= Google Wormdingler`

`cat /proc/device-tree/model= Google Wormdingler`

`xinput list
⎡ Virtual core pointer                          id=2    [master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer                id=4    [slave  pointer  (2)]
⎜   ↳ hid-over-i2c 0603:604A                    id=6    [slave  pointer  (2)]
⎜   ↳ hid-over-i2c 0603:604A Stylus stylus      id=7    [slave  pointer  (2)]
⎜   ↳ Google Inc. Hammer                        id=9    [slave  pointer  (2)]
⎜   ↳ hid-over-i2c 0603:604A Stylus eraser      id=12   [slave  pointer  (2)]
⎜   ↳ hid-over-i2c 0603:604A Stylus pad         id=13   [slave  pointer  (2)]
⎣ Virtual core keyboard                         id=3    [master keyboard (2)]
    ↳ Virtual core XTEST keyboard               id=5    [slave  keyboard (3)]
    ↳ Google Inc. Hammer                        id=8    [slave  keyboard (3)]
    ↳ cros_ec_buttons                           id=10   [slave  keyboard (3)]
    ↳ sc7180-rt5682-max98357a-1mic Headset Jack id=11   [slave  keyboard (3)]`

archisman-panigrahi, 2025-05-18:
@WesleyVanNeck most likely your stylus input is `hid-over-i2c 0603:604A Stylus stylus` and the touchscreen input is `hid-over-i2c 0603:604A` (or, is it Google Inc. Hammer?) Could you verify if these inputs work in the screen rotation manager app (only works on x11)

WesleyVanNeck, 2025-05-18:
> [@WesleyVanNeck](https://github.com/WesleyVanNeck) most likely your stylus input is `hid-over-i2c 0603:604A Stylus stylus` and the touchscreen input is `hid-over-i2c 0603:604A` (or, is it Google [...]

wich app i dont fully understand

archisman-panigrahi, 2025-05-18:
Yes, this app - it is available on the velvet os via a repo https://github.com/velvet-os/velvet-os.github.io/blob/main/postinst/touch-input-rotate.md#appropriately-rotate-touchscreen-input-and-touchpad-input-while-rotating-screen-in-xorg

WesleyVanNeck, 2025-05-18:
> [@WesleyVanNeck](https://github.com/WesleyVanNeck) most likely your stylus input is `hid-over-i2c 0603:604A Stylus stylus` and the touchscreen input is `hid-over-i2c 0603:604A` (or, is it Google [...]

thats correct Google Inc. Hammer is the touchpad

archisman-panigrahi, 2025-05-18:
So your chromebook has all three of touchscreen, touchpad and stylus?

In that case I will have to upgrade my app to take three inputs instead of two

archisman-panigrahi, 2025-05-24:
@hexdump Could you update your top comment with the fact that we also need output of `cat /proc/device-tree/model`?

hexdump0815, 2025-05-24:
@archisman-panigrahi - added

archisman-panigrahi, 2025-05-27:
@WesleyVanNeck I pushed a new update (v1.3.1) which has a autodetect support for your model, and also has stylus support (in addition to touchscreen and touchpad). It is already on the velvet-os repo, you can install it with `sudo apt install screen-orientation-manager`. I'd be happy to know your feedback. If there is any issue, please open an issue in my app's github issues.

archisman-panigrahi, 2025-08-06:
@pyrosrock If this is not a chromebook, please open an issue here https://github.com/archisman-panigrahi/surface-RT-screen-rotator/issues/ and post the output of `xinput list` while running X11 (not wayland)