velvet-os/imagebuilder #340: Is there any way to boot the image off SD card in Amazon Austin without replacing lineageOS?

archisman-panigrahi, 2025-05-26:
I have an Amazon Austin tablet running LineageOS (installed in internal memory). 

Is there any way I can boot this image (https://github.com/velvet-os/imagebuilder/blob/main/systems/tablet_amazon_austin/readme.md) from a micro SD card without removing the LineageOS in the internal memory?

Or, is there some way I can dual boot? (LineageOS from internal memory, Debian from SD Card)

hexdump0815, 2025-05-26:
i think there is no way to dual boot - the bootloader is that limited and locked that you already have to do a lot of extra stuff to get anything not fire-os booted at all with a lot of hacks that there is close to no way to get it to doing more complex things like dual booting etc. ... maybe just keep the lineageos as the kernel from the linux images is so old and limited that it does not really make much sense to run linux on it - i also do not plan to really work on those amazon tablets anymore ass the hardware is simply not good for open source - chromebooks and sometimes tv boxes are way better in this regard

archisman-panigrahi, 2025-05-26:
Hmm, the performance as an ebook reader is very good with LineageOS 14.1 + koreader. Maybe I should just stick to that.

Anyway, thank you for responding, and thank you for this project.