velvet-os/imagebuilder #344: Wifi won't work on CB5-311 after an apt update

cylin577, 2025-06-13:
I'm using the image from [here](https://github.com/velvet-os/imagebuilder/releases/tag/230915-02).The problem is that it can't find any network device

cylin577, 2025-06-13:
I tried the image that you provided,The xfce Wifi GUI thing will still just say "No network device aviliable"

cylin577, 2025-06-14:
[dmesg.log](https://github.com/user-attachments/files/20735563/dmesg.log)
Here it is

hexdump0815, 2025-06-15:
it looks like there is a problem initializing the wifi card:
```
mmc2: error -110 whilst initialising SDIO card
mmc2: Failed to initialize a non-removable card
```
i'll check this again on my nyan soon - did(does wifi work on that system in chromeos?

hexdump0815, 2025-06-15:
it probes well on my nyan - it looks a bit like a hardware problem on yours or different hardware variations across the nyan devices ... one option might be to reduce the sdio speed and/or frequency in the dts/dtb and see if that helps ... i might try to create a dtb for you with reduces values, but it might take a while - please remind me here if you do not hear from me within the next two weeks or make yourself familiar with decompiling/changing dtb/dts files

worst case you should be able to use some usb wifi dongle

cylin577, 2025-06-15:
The Wifi feature works well on chromeos and I already bought a USB Wifi dongle but Linux doesn't support it yet

hexdump0815, 2025-06-17:
strange - could you please maybe try some older nyan images to see if the problem exists only with the newer ones (i.e. is some regression)?

cylin577, 2025-06-18:
I tried 80% of the images  but none of them works

hexdump0815, 2025-06-18:
ok - then something around your hardware seems to be different to all the systems i (and others) used this with ... as said, i'll try to prepare a dtb with reduced speed to try later - not sure if that will help though

hexdump0815, 2025-07-27:
thanks for the reminder - i'll soon build a new set of kernels and will do it as part of that