velvet-os/imagebuilder #345: lenovo duet 5 on kernal 6.12 cant get qcom drivers

theotherside101, 2025-06-15:
i get the following error.

sudo apt-get -y install qrtr-tools rmtfs
[sudo] password for linux: 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package qrtr-tools
E: Unable to locate package rmtfs
linux@changeme:~$

thenameisluk, 2025-06-16:
i've checked and these packages aren't available in ubuntu jammy anymore

if you want your wifi to work either

use debian

or

get .deb package from ppa or other ubuntu version that has it

thenameisluk, 2025-06-16:
here is a ppa that should work https://launchpad.net/~liujianfeng1994/+archive/ubuntu/qcom-mainline

theotherside101, 2025-06-17:
I tried Debian and it had wifi but now it won't install I've ran it 5 times heck even ubuntu doesn't work I've tried the manual one and the one that is automatic it doesn't work I can reinstall chrome os it boots fine but ubuntu/debian won't work

thenameisluk, 2025-06-17:
very cool 
i ack that it isn't working 

but unless you provide :
- logs from the installation script
- velvet tools version

i won't be helping you because i'm not psychic

Sanketpathania, 2025-07-06:
Hey i am using the latest vtinstall script but after installation and rebooting it never boots and starts a bootloop as soon as I click on boot from internal disk - it install fine thru the whole process but at the last it says no kernel specified going with current kernel 6.1 and rm: cannot remove mnt/boot/vmlinux.kpart-initrd-* no such file or directory 

am i missing a step in my installtion - did a full upgrade too

thenameisluk, 2025-07-07:
@Sanketpathania 
i would recommend you using the latest testing image

it works very well there after full upgrade
https://github.com/velvet-os/imagebuilder-testimages/releases/tag/spring-release-2025-testimages-round-01

Sanketpathania, 2025-07-07:
> [@Sanketpathania](https://github.com/Sanketpathania) i would recommend you using the latest testing image
> 
> it works very well there after full upgrade https://github.com/velvet-os/imagebuilder-testimages/releases/tag/spring-release-2025-testimages-round-01

Thank you very much!!!
Will install today and advise how it went.

archisman-panigrahi, 2025-09-03:
This issue can be closed