velvet-os/imagebuilder #353: GBB Flag Commands

CodingGenius0001, 2025-07-18:
Hello @thenameisluk @hexdump0815!

As referenced in this issue [https://github.com/velvet-os/velvet-tools/issues/8], the new GBB flags are different for newer chromebooks and so the following commands no longer work on newer versions of ChromeOS as they error out as invalid or incomplete commands.

`/usr/share/vboot/bin/set_gbb_flags.sh 0x19`

`futility gbb --set --flash --flags=0x19`

`futility gbb --get --flash --flags`

Please suggest some updated ones to use in order to set gbb flags!

Thanks!

thenameisluk, 2025-07-21:
i think you might be experiencing this issue https://github.com/velvet-os/imagebuilder/issues/130#issuecomment-2566571648

thenameisluk, 2025-07-21:
or you haven't disabled write protection

thenameisluk, 2025-07-21:
very well then pls share full command output of all instead of stating that it doesn't work thank you

CodingGenius0001, 2025-07-21:
@thenameisluk ok so i have been doing some tinkering around. this command finally worked for me. flag 0x19 is still valid just that the command changed slightly.

old command: `futility gbb --set --flash --flags=0x19`

new command: `futility gbb --set --flags=0x19 --flash`

hope this helped!

thenameisluk, 2025-07-21:
that's weird in theory the arguments should be position independet

but very well i'll note that in the documentation

closed 2025-07-22 by CodingGenius0001

hexdump0815, 2025-07-23:
i'm also surprised that the order should play any role, but if it helps it will be fine :)

CodingGenius0001, 2025-08-03:
cd ~
sudo flashrom -p linux_mtd:dev=0 -r firmware-backup.bin
ls -lh firmware-backup.bin

---

gbb_utility --get --flags firmware-backup.bin

---

gbb_utility --set --flags 0x0 \
  firmware-backup.bin \
  firmware-zero-flags.bin
ls -lh firmware-zero-flags.bin

---

sudo flashrom -p linux_mtd:dev=0 \
  -w firmware-zero-flags.bin

---

sudo reboot
# After login, dump just the GBB region:
sudo flashrom -p linux_mtd:dev=0 -i GBB:current-gbb.bin
gbb_utility --get --flags current-gbb.bin

@thenameisluk @hexdump0815 

please put this in the docs as i believe this is very beneficial for people trying to switch off gbb flags again

CodingGenius0001, 2025-08-03:
from my testing these are the commands that successfully allow anyone to set the gbb flags back to 0x0

`cd ~
sudo flashrom -p linux_mtd:dev=0 -r firmware-backup.bin
ls -lh firmware-backup.bin`

---

`gbb_utility --get --flags firmware-backup.bin`

---

`gbb_utility --set --flags 0x0 \
  firmware-backup.bin \
  firmware-zero-flags.bin
ls -lh firmware-zero-flags.bin`

---

`sudo flashrom -p linux_mtd:dev=0 \
  -w firmware-zero-flags.bin`

---

`sudo reboot
# After login, dump just the GBB region:
sudo flashrom -p linux_mtd:dev=0 -i GBB:current-gbb.bin
gbb_utility --get --flags current-gbb.bin`

---

`cd ~
sudo flashrom -p linux_mtd:dev=0 -r firmware-postflash.bin
gbb_utility --get --flags firmware-postflash.bin`