velvet-os/imagebuilder #355: Question

yicheng42, 2025-07-22:
对于Lenovo 10e(kukui)的弹簧发布镜像(spring-release-test01)，还有什么bug存在吗？
并且我应如何使用vtinstall.txt?(https://github.com/velvet-os/velvet-tools/blob/main/scripts/vtinstall)
麻烦能不能给个详细的说明？
 Please translate into English...

thenameisluk, 2025-07-22:
@hexdump0815 the guy is asking about bugs in spring-release-test01 where there any

@yicheng42 regarding the usage of vtinstall, this one is used to copy over velvet os file system to internal memory

assumming the distro you are using has same partition structure as ours it should work for you

thenameisluk, 2025-07-23:
@yicheng42 if you want to actually build an image not just copy it please refer to https://github.com/velvet-os/velvet-os.github.io/blob/main/using-the-imagebuilder-framework.md
all required files should be in this repo