velvet-os/imagebuilder #362: [kukui-cozmo-trixie]  error while installing wireguard & not able to start interface

gallee, 2025-08-07:
Hello, 

When trying to install wireguard on trixie, I have and the following error running on 6.12.28-stb-cbm+ / trixie

```
$ sudo apt install wireguard
Installing:                     
  wireguard

Installing dependencies:
  linux-image-6.12.38+deb13-rt-arm64  linux-image-rt-arm64  wireguard-tools

Suggested packages:
  linux-doc-6.12  debian-kernel-handbook

Summary:
  Upgrading: 0, Installing: 4, Removing: 0, Not Upgrading: 0
  Download size: 94.0 MB
  Space needed: 179 MB / 310 MB available
  └─ in /boot:  26.1 MB / 383 MB available

Continue? [Y/n] Y
Get:1 http://deb.debian.org/debian trixie/main arm64 linux-image-6.12.38+deb13-rt-arm64 arm64 6.12.38-1 [93.9 MB]
Get:2 http://deb.debian.org/debian trixie/main arm64 linux-image-rt-arm64 arm64 6.12.38-1 [1,444 B]                                                                                          
Get:3 http://deb.debian.org/debian trixie/main arm64 wireguard-tools arm64 1.0.20210914-3 [86.6 kB]                                                                                          
Get:4 http://deb.debian.org/debian trixie/main arm64 wireguard all 1.0.20210914-3 [5,804 B]                                                                                                  
Fetched 94.0 MB in 30s (3,136 kB/s)                                                                                                                                                          
Selecting previously unselected package linux-image-6.12.38+deb13-rt-arm64.
(Reading database ... 98436 files and directories currently installed.)
Preparing to unpack .../linux-image-6.12.38+deb13-rt-arm64_6.12.38-1_arm64.deb ...
Unpacking linux-image-6.12.38+deb13-rt-arm64 (6.12.38-1) ...
Selecting previously unselected package linux-image-rt-arm64.
Preparing to unpack .../linux-image-rt-arm64_6.12.38-1_arm64.deb ...
Unpacking linux-image-rt-arm64 (6.12.38-1) ...
Selecting previously unselected package wireguard-tools.
Preparing to unpack .../wireguard-tools_1.0.20210914-3_arm64.deb ...
Unpacking wireguard-tools (1.0.20210914-3) ...
Selecting previously unselected package wireguard.
Preparing to unpack .../wireguard_1.0.20210914-3_all.deb ...
Unpacking wireguard (1.0.20210914-3) ...
Setting up linux-image-6.12.38+deb13-rt-arm64 (6.12.38-1) ...
I: /vmlinuz.old is now a symlink to boot/vmlinuz-6.12.38+deb13-rt-arm64
I: /initrd.img.old is now a symlink to boot/initrd.img-6.12.38+deb13-rt-arm64
I: /vmlinuz is now a symlink to boot/vmlinuz-6.12.38+deb13-rt-arm64
I: /initrd.img is now a symlink to boot/initrd.img-6.12.38+deb13-rt-arm64
/etc/kernel/postinst.d/initramfs-tools:
update-initramfs: Generating /boot/initrd.img-6.12.38+deb13-rt-arm64
Image-6.12.38+deb13-rt-arm64 seams to be missing
ple ple ple your kernel version is likely wrong
WARNING. something went wrong while building the image pls be carefull while flashing it
run-parts: /etc/initramfs/post-update.d//velvet-build exited with return code 1
run-parts: /etc/kernel/postinst.d/initramfs-tools exited with return code 1
dpkg: error processing package linux-image-6.12.38+deb13-rt-arm64 (--configure):
 installed linux-image-6.12.38+deb13-rt-arm64 package post-installation script subprocess returned error exit status 1
Setting up wireguard-tools (1.0.20210914-3) ...
wg-quick.target is a disabled or a static unit not running, not starting it.
dpkg: dependency problems prevent configuration of linux-image-rt-arm64:
 linux-image-rt-arm64 depends on linux-image-6.12.38+deb13-rt-arm64 (= 6.12.38-1); however:
  Package linux-image-6.12.38+deb13-rt-arm64 is not configured yet.

dpkg: error processing package linux-image-rt-arm64 (--configure):
 dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of wireguard:
 wireguard depends on wireguard-modules (>= 0.0.20191219) | wireguard-dkms (>= 0.0.20200121-2); however:
  Package wireguard-modules is not installed.
  Package linux-image-rt-arm64 which provides wireguard-modules is not configured yet.
  Package wireguard-dkms is not installed.

dpkg: error processing package wireguard (--configure):
 dependency problems - leaving unconfigured
Processing triggers for man-db (2.13.1-1) ...
Errors were encountered while processing:
 linux-image-6.12.38+deb13-rt-arm64
 linux-image-rt-arm64
 wireguard
Error: Sub-process /usr/bin/dpkg returned an error code (1)

```

Then when I launch wireguard via `sudo systemctl start wg-quick@wg0.service` 

```
Aug 07 10:31:38 cozmo wg-quick[7040]: [#] nft -f /dev/fd/63
Aug 07 10:31:39 cozmo wg-quick[7115]: /dev/fd/63:5:68-81: Error: Could not process rule: No such file or directory
```

This may be related to some kernel parameters not set see [here(https://github.com/ViRb3/wgcf/issues/55#issuecomment-2346510243)

```
cat /boot/config-6.12.28-stb-cbm+ | grep CONFIG_NFT_FIB_IPV4
# CONFIG_NFT_FIB_IPV4 is not set
```

hexdump0815, 2025-08-09:
i added the options so that future kernels will have them: https://github.com/hexdump0815/kernel-config-options/commit/bb107fc655af7578021776a840a2aca7ce8f4971

hexdump0815, 2025-08-09:
@thenameisluk - any idea if that might be related? i guess its just triggering an initramfs rebuild which then fails due to the debian wireguard being tightly pacakged with the kernel we do not use (i think), so not the real root cause ... but this is just a guess

thenameisluk, 2025-08-11:
```
Image-6.12.38+deb13-rt-arm64 seams to be missing
ple ple ple your kernel version is likely wrong
WARNING. something went wrong while building the image pls be carefull while flashing it
run-parts: /etc/initramfs/post-update.d//velvet-build exited with return code 1
run-parts: /etc/kernel/postinst.d/initramfs-tools exited with return code 1
```

what i see is debian attempts to pass their kernel to vt and it failing
baceuse the diffrent naming convention ig, still it didn't got flashed so i doubt it's the issue

i would't try to flash it anyways since it's not patched to support chromebooks

hexdump0815, 2025-08-11:
oh - so the debian kernel might get pulled in by dependencies of the wireguard stuff ... looks like at some point we might have to look closer at how that is supposed to work - maybe a wireguard userland install without its deps is possible? i think we have the wireguard module and options by default in our kernels (at least in future ones), so enabling wiregued should be a userspace topic only for us - but we will have to see how to combine it cleanly with how wireguard is packaged in debian now