velvet-os/imagebuilder #366: help!

virajbharti15-design, 2025-09-03:
hey hexdump0815 i need your help with something on the ubuntu 22.04 i dont know how to use github im really sorry but i just need help with downloading the kukui chromebook something onto the chromebook itself so i can boot it without the usb please help

archisman-panigrahi, 2025-09-03:
> something onto the chromebook itself so i can boot it without the usb

You can also use a microSD card to boot. Once you have a bootable image then you can flash it to the eMMC so that you will not need the SD card/USB drive.

virajbharti15-design, 2025-09-03:
debian trixie works horribly slow on my chromebook... is there really no way to download ubuntu from the usb? @hexdump0815

archisman-panigrahi, 2025-09-03:
> debian trixie works horribly slow on my chromebook... is there really no way to download ubuntu from the usb?

Usually nearly all the operating systems are horribly slow when running from a USB drive/sdcard. But they become fast after you flash them to the internal storage.

You can also download the ubuntu image in the USB while running Debian from USB, and flash it to the internal memory, without loading it to the USB.

Can you please clarify what is the current state of installation and what exactly you are trying to achieve?

virajbharti15-design, 2025-09-04:
@archisman-panigrahi thanks for the reply

im stupid, just straight up dumb, i dont know what the hell im doing, all i know is that i got ubuntu jammy working on my lenovo ip flex 3 chromebook, and i tried debian whatever on it earlier and this runs a hell of a lot better, my problem is, i need the usb to run ubuntu. i want to install ubuntu to internal storage or whatever its called so i can just press ctrl U from the developer mode screen and it runs WITHOUT the usb

archisman-panigrahi, 2025-09-04:
To install to emmc, follow the steps in https://velvet-os.github.io/chromebooks/installation/basic-installation.html, and ask us if there is any confusion

virajbharti15-design, 2025-09-20:
@archisman-panigrahi it doesnt boot, most likely from that unmounting error

virajbharti15-design, 2025-09-20:
@archisman-panigrahi i figured it out, the "/mnt is busy" doesn't mean anything as long as a disk pops up somewhere on the screen, i also had to install it to mmcblk0 because my chromebook (Lenovo IP Flex 3) will not read any other drive

I also figured out how to replace xfce with gnome, all i did was uninstall the xfce files

sudo apt purge xfce4 xfce4-*
sudo apt purge lightdm

then manually installed gnome packages

sudo apt install gnome-session gnome-shell gdm3 gnome-control-center gnome-tweaks
sudo apt install gnome-shell-extensions gnome-settings-daemon gnome-backgrounds gnome-screensaver
sudo apt install gnome-terminal gnome-system-monitor gnome-calculator gnome-disk-utility

replacing xfce with gnome can be done while it is being booted from the USB, and then you can follow the installation procedure, making it feel like youre literally installing ubuntu on your chromebook

archisman-panigrahi, 2025-09-20:
Please close the issue if the error is fixed

closed 2025-09-20 by virajbharti15-design