velvet-os/imagebuilder #367: nyan: Can't boot from USB

cylin577, 2025-09-03:
So I have already installed bookworm to the emmc but I messed the uboot configuration up so now I need to boot from a usb which contains a clean image but it will never boot from the USB no matter what, I reimaged it abt 5 times, NOTHING, I imaged trixie onto it, NOTHING, I tried focal, and you guessed it, NOTHING.

cylin577, 2025-09-03:
Also, my nyan is werid, normaly, some nyan have a 1366x768 screen w/ 2GB of ram (According to the documents), but mine is 1366x768 screen w/ 4GB of ram

cylin577, 2025-09-03:
I havent tried PMos so let me try that

hexdump0815, 2025-09-03:
@cylin577 - it could be that all possible combinations of ram and screen exist for nyan systems - i only mentioned the ones i have seen or heared about

your booting problem might be because u-boot reads the extlinux.conf on the emmc first and if that is broken it will not properly boot further from usb/sd - i have that as a todo to change the extlinux.conf reading order in the u-boot build ... one thing which might work is to dd the vmlinux kpart image in the /boot (third) partition to the first partition - then u-boot is not used and you should be able to boot and from the booted system you can fix or move away your broken extlinux.conf ... please report back here if that worked

archisman-panigrahi, 2025-09-03:
> So I have already installed bookworm to the emmc but I messed the uboot configuration

DId you dd the image into emmc?

From my experience, if you already dd the image to emmc, it will not boot from USB (flashed with Debian) after that. You have to restore chromeOS using another USB, and then boot Debian from USB and install to emmc.

cylin577, 2025-09-05:
I will close the issue bc I fixed it using pmos and I also figured out how to install packages

closed 2025-09-05 by cylin577